udacity/robotics-beta-test-issues #25: diyjac L01E03

diyjac, 2017-04-18:
In setting up your Environment, the github repository readme should have more details on the setting up the environment, instead of the link `here`.  Reading this makes me think that it was only for downloading and setting Anaconda and not the complete environment.

UsmanIjaz, 2017-04-18:
Isn't it L02E03 ?

diyjac, 2017-04-19:
You are right, I am skipping Welcome as L01, so should be L02.

ryan-keenan, 2017-04-22:
Thanks John, I've changed this link for now to the [configure and manage your environment with anaconda](https://github.com/ryan-keenan/RoboND-Python-Starterkit/blob/master/doc/configure_via_anaconda.md) link.  Will be cleaning this up more in weeks to come as well ;)

closed 2017-04-22 by ryan-keenan