udacity/robotics-beta-test-issues #30: Usman L02E04

UsmanIjaz, 2017-04-18:
A video for data visualization would be very helpful, describing the different columns. Reading demands much more effort then watching a video. Video explains a lot more then reading. It was done in SDCND:
https://classroom.udacity.com/nanodegrees/nd013/parts/fbf77062-5703-404e-b60c-95b78b2f3f9e/modules/6df7ae49-c61c-4bb2-a23e-6527e69209ec/lessons/46a70500-493e-4057-a78e-b3075933709d/concepts/4b26b52d-8eed-4b04-91a6-f1d421bb04f1

ryan-keenan, 2017-04-22:
Will do this.  Need to wait for the simulator environment and output to stabilize a bit more before recording so as not to have to re-do

ryan-keenan, 2017-05-09:
Have recorded a video on this and it will be in the classroom soon!

ryan-keenan, 2017-05-22:
video now in the classroom!

closed 2017-05-22 by ryan-keenan