udacity/robotics-beta-test-issues #31: Usman L02E05

UsmanIjaz, 2017-04-18:
"Color thresholding" should be divided into
-The Perception Step
-Color Thresholding
- Quiz/Exercise
Firstly, three things are discussed under one part. 
Secondly, too much information in one part. The page has also become very long. It would be problematic for students to revisit a particular topic.

ryan-keenan, 2017-04-22:
Cool, I'll look at breaking this up.

ryan-keenan, 2017-05-04:
Thanks for all this great feedback! I've addressed  some of this now. At this point, two sub-heading now exist "Reading in Color Images" and "Color Thresholding" and the name of the page has been switched to "The Perception Step". Also added another image above the quiz to make it clear what the output should look like.  I will keep an open mind to breaking this into two parts but for now leave it as one because another issue we're trying to keep in mind is total number of concepts (pages) in the lesson.  For now we tend to err on the side of letting a concept go longer rather than parsing things out.

closed 2017-05-04 by ryan-keenan