udacity/robotics-beta-test-issues #42: Usman L02E06

UsmanIjaz, 2017-04-19:
There is a lot of text information on this page. It should be divided in atleast four parts. 
-Video explaining perspective transform, include the description at the bottom of the page
-Quiz to test the real intuition of Perspective transform
-A video explaining the code and one example like perspective transform of stop sign video in SDCND.
-A quiz at the end to select source and destination points.

ryan-keenan, 2017-04-22:
Thanks, I"ll look into breaking this up.

ryan-keenan, 2017-05-22:
Great suggestions!  I have added a video here with visual descriptions and a code walkthrough.  Everything is still on one page but I think it's much more clear and approachable now.

closed 2017-05-22 by ryan-keenan