udacity/robotics-beta-test-issues #55: letyrodri L01E011

letyrodridc, 2017-04-19:
Jupyter notebook image (screenshot) in lesson doesn't correspond to notebook current version.

KvalheimRacing, 2017-04-25:
Isen't this because of the random image choosing?

idx = np.random.randint(0, len(img_list)-1)
image = mpimg.imread(img_list[idx])

letyrodridc, 2017-04-26:
? This post isn't talking about code. It's just the image included in the classroom lesson.

KvalheimRacing, 2017-05-01:
I though that you meant your that the picture in your notebook is not the same as the one in the lesson.

ryan-keenan, 2017-05-09:
Thanks for the comment here!  I'm not going to change this right now but will keep an eye out for inconsistencies (did add a comment in the lesson that your notebook should look "similar" to the one pictured above).

closed 2017-05-09 by ryan-keenan