udacity/robotics-beta-test-issues #97: P01E06

stupsi99, 2017-04-19:


stupsi99, 2017-04-19:
It took me a long time to understand where the matrix entries from the destination matrix were coming from. I think a small image of what the points actually are would have been great! Also, I would suggest to comment in the code where the "dst_size" value is coming from.

stupsi99, 2017-04-20:
Actually the very first image on that page is already quite good. Combine the transformed checkers board with a coordinate system and its clear how to get to the destination points. Also placing the original image above the transformed image would help (instead of having them side by side).

ryan-keenan, 2017-04-23:
I've reorganized things on this page a bit to hopefully be more understandable.   We'll move forward now with making a video for this concept to help visualize things better.  Thanks for your suggestions here!

closed 2017-04-23 by ryan-keenan