udacity/robotics-beta-test-issues #98: letyrodri P1

letyrodridc, 2017-04-19:
It's not clear in perception.py:

1) what Rover.vision_img has to be. I used perpect_transform, but it seems that could be an image with more analysis. 

2) Not clear any of these:
 
# Compute yaw values and x,y positions of navigable pixels in world space
    # Update Rover worldmap

I have just updated the worldmap with the rover centric map (visual space of the rover), like is shown in L01E08, but it's not clear if that was expected.

letyrodridc, 2017-04-20:
This issue could be close. A video of the rover was added in the Beta - E01 Overview that clarifies this points.

ryan-keenan, 2017-05-01:
Thanks for this feedback!  Glad to hear the new content helps :)

closed 2017-05-01 by ryan-keenan