udacity/robotics-beta-test-issues #99: letyrodri P1

letyrodridc, 2017-04-19:
I don't see clear in the project, what analysis is expected over the images, and which actions could be taken by the rover. 

For example, is it expected to the Rover turn when he hit a rock?

The project seems open to ideas but could be a minimum of actions expected just as a guide.

Maybe, indicates: the autonomous rover should do "these actions" to pass the project and implement some new idea.

letyrodridc, 2017-04-23:
I've found information about what is expected in writeup_template.md . This post could be related to: https://github.com/udacity/robotics-beta-test-issues/issues/123

letyrodridc, 2017-04-24:
This issue could be closed. The Rubric was added to Beta Version.

letyrodridc, 2017-04-24:
This issue could be closed. The rubric was added to beta test version.

ryan-keenan, 2017-05-01:
Thanks for the feedback!  Yes I've been adding things to help with these issues :)

closed 2017-05-01 by ryan-keenan