udacity/robotics-beta-test-issues #123: letyrodri P1

letyrodridc, 2017-04-21:
In writeup_template.md, there is information about what is expected to the student achieve. I've read it at the end, after I finished beta testing the project. In classroom, it's not indicated clearly what the student has to do. Those indication could be moved from writeup_template.md to classroom or readme.md as AIND does.

letyrodridc, 2017-04-23:
Or, it can be a clear indication in classroom : The project instructions are in writeup_template.md . I'm guessing that maybe there is easy to modify by project developer.

ryan-keenan, 2017-04-23:
Thanks for the feedback!  The instructions are more clearly laid out in the [project rubric](https://review.udacity.com/#!/rubrics/916/view) now, which is referenced in the classroom project description.  I will also be adding more explicit instructions in the lesson.

closed 2017-04-23 by ryan-keenan