udacity/robotics-beta-test-issues #139: EirikKvalheim L01E07

KvalheimRacing, 2017-04-23:
I think it would better help the overall understanding, if the function rover_coords(binary_img) takes in camera postion like this:

# Getting coordinates with respect to the rover

def rover_coords(binary_img, camera_pos):
    
    ypos, xpos = binary_img.nonzero()

    x_pixel = (xpos - camera_pos[0])
    y_pixel = np.absolute(ypos - camera_pos[1])

    return x_pixel, y_pixel

And then running it like this:

camera_pos = [160,160] # x and y values for where the camera is positioned
xpix, ypix = rover_coords(colorsel, camera_pos)
plt.plot(xpix, ypix, '.')

Because here it randomly occured that the length of the y axis on the image had the same coords as the x,y pos of the camera. This is not a good way to do it/or expect students to do it, because pictures can be of different sizes.

KvalheimRacing, 2017-04-23:
Sorry for wrong formatting, just copyed code into here, and it went bananas on the formatting x)

ryan-keenan, 2017-04-26:
Good point, it's a bit lazy on my part to just use `image.shape[0]` in both conversions because that relies on the 1x2 aspect ratio.  For this use case though we always expect the camera to be at the bottom center so how about this version:
```
def rover_coords(binary_img):
    ypos, xpos = binary_img.nonzero()
    # Calculate pixel positions with reference to the rover position being at the 
    # center bottom of the image.  
    y_pixel = np.absolute(ypos - binary_img.shape[0])
    x_pixel = (xpos - np.int(binary_img.shape[1] / 2))
    return x_pixel, y_pixel
```

ryan-keenan, 2017-05-01:
ok have converted the solution to the above suggestion

closed 2017-05-01 by ryan-keenan

KvalheimRacing, 2017-05-01:
Yeah, It is ok as long as you state that you " always expect the camera to be at the bottom center".