udacity/robotics-beta-test-issues #141: Eirik_Kvalheim P01

KvalheimRacing, 2017-04-24:
Trying to get an overview over what's needed on P1, I'm getting very confused. Right now I am looking in; "Navigating the Environment", "Rover Project Test Notebook.ipynb", Project Readme, Project writeup_template, and "Project Submission", all these places there seems to be info on what one is supposed to do, but not the same info ( a little different everywhere). This require people to use time on understanding what to do, rather than doing it.

My suggestion is to have *one* place with the project requirements.

KvalheimRacing, 2017-04-24:
There is also ofc info/requirements in the rubic. So in total 6 places to look.

ryan-keenan, 2017-05-09:
Thanks for this feedback, things were definitely a bit scattered.  I've cleaned this up somewhat so that now the details are mostly in the lesson, and the "Project Submission" page links back to relevant details.  The project repo is intended to be a standalone thing so some repetition there is necessary and the rubric is required for the reviews process, so some repetition is required there as well.  All in all we like redundancy because some people will miss certain details if they don't appear in more than one place, but we certainly don't aim to add more confusion than necessary!  Hopefully it's in better shape now.

closed 2017-05-09 by ryan-keenan