udacity/robotics-beta-test-issues #142: Eirik_Kvalheim P01

KvalheimRacing, 2017-04-25:
When making the video from processed image data in the Jupyter notebook, you originally make a `img_list_sorted` straight out of the csv file. Here the algorithms asumes that everything is formatted correct. I had recorded some different runs in the simulator, and had the header `Path,SteerAngle,Throttle,Brake,Speed,X_Position,Y_Position,Pitch,Yaw,Roll` displayed several times in the log file. Then the algorithm crashes because `Path` is not a valid directory, obv. 
I assume more people than me will do this "mistake", and you could fix it very easy with something like this:

`df = pd.read_csv('../../robot_log.csv')

img_list_sorted = [] # Create list of image pathnames

for i in df["Path"].tolist():
    if len(i) > 5:
        img_list_sorted.append(i)`

KvalheimRacing, 2017-04-25:
Sorry, wrong formatting there. Meant this;

```
df = pd.read_csv('../../robot_log.csv')

img_list_sorted = [] # Create list of image pathnames

for i in df["Path"].tolist():
        if len(i) > 5:
              img_list_sorted.append(i)```

ryan-keenan, 2017-05-02:
Thanks for pointing this out!  The simulator is now set to clear the old IMG folder and csv file before recording new data.  While I realize this may cause other problems of wiping out data you might have wanted to keep, it seems like a clean solution for now and I'll stick in a warning somewhere that you need to rename your data folder/csv if you want it to not be overwritten.

closed 2017-05-02 by ryan-keenan