udacity/robotics-beta-test-issues #143: Eirik_Kvalheim P01

KvalheimRacing, 2017-04-25:
Simulator pickup not working in traning mode (havent tried autonomous mode).
pressing the letters for pickup, tried together, after each other, and bith with caps on/off. Nothing happens, and I am in reach, because it stands "Yes" on is near object.

Reset zoom doesent work either, so seems the letter combinations is not working for commands. Can it have something to do with different kind of keyboards? Mine is not US..

ryan-keenan, 2017-04-25:
Interesting, did you try left mouse button for pickup?  Could be a keyboard issue

ryan-keenan, 2017-05-22:
On startup now you can click the "input" tab on the screen that allows you to choose resolution and graphics quality to change keyboard inputs.  If you're still having issues please give that at try.

closed 2017-05-22 by ryan-keenan