udacity/robotics-beta-test-issues #146: yazeedalrubyli L01E05

yazeedalrubyli, 2017-04-25:
The video is good but it is also general. Anyone familiar with any ND will definitely pass by this video. People spend time and money for the ND to eventually land on their dream jobs. So, this important to focus on and make a new video focusing on Robotics and partnerships that Udacity has to empower their students to find their dream job. It is a waste of resources if Udacity did not talk about their partnerships with leading company like Mercedes-Benz, NVIDIA and more in benefits of students.

yazeedalrubyli, 2017-04-25:
Take SDCND's Career Video as​ a baseline.

ryan-keenan, 2017-05-02:
Thanks for this suggestion!  We are working on more content with partners for the careers sections so this will be forthcoming.

ryan-keenan, 2017-05-22:
We have some great video content now for this type of thing from the livestreams we've done, just need to cut it and get it in the classroom.  Hopefully by wednesday!

ryan-keenan, 2017-05-27:
got our first "careers in robotics" video in there now, and more to come!

closed 2017-05-27 by ryan-keenan