udacity/robotics-beta-test-issues #147: yazeedalrubyli P1E03

yazeedalrubyli, 2017-04-25:
'We'll be using Anaconda from continuum analytics to install Python and Jupyter Notebooks to do *analysis in the projects*. '

Notebook is not only for analysis.

ryan-keenan, 2017-04-28:
I'm not sure exactly what you mean here... could you elaborate?

yazeedalrubyli, 2017-04-30:
Do we use Python and Jupyter Notebooks to do (analysis), not a complete project? If we do more than analysis, we should say `to do the projects. ` instead of `to do analysis in the projects. `

ryan-keenan, 2017-05-01:
Ah I see your point.  For the most part, we'll be using notebooks for analysis but not as the final result.   In this project it's sort of half and half, some of the project is in the notebook but the main result is using the autonomous driving scripts with the simulator.

closed 2017-05-01 by ryan-keenan