udacity/robotics-beta-test-issues #148: yazeedalrubyli P01

yazeedalrubyli, 2017-04-25:
I don't know if you're aware of this, but the picker in the simulator is glitchy and controlling it to move around is hard.

yazeedalrubyli, 2017-04-25:
macOS

ryan-keenan, 2017-04-28:
Thanks for reporting this.  If you could say exactly how you're having trouble I'll try to replicate on my mac

yazeedalrubyli, 2017-04-30:
Going forward then backward I have to press three/four times to start moving forward again.

ryan-keenan, 2017-05-01:
I'll look into this, but I haven't seen this on my own mac

yazeedalrubyli, 2017-05-01:
I tried it again and I figured out the problem, if I turn with a forward move but I click turn button first it will stuck. I understand that now because of the circle movement around itself when only press turns buttons (Left/Right).​

ryan-keenan, 2017-05-04:
Ok I'm going to close this issue for now but if some issue remains feel free to reopen or create a new issue

closed 2017-05-04 by ryan-keenan