udacity/robotics-beta-test-issues #153: sychaichangkun L01E05

sychaichangkun, 2017-04-26:
The text is a bit long for learners and there is only one second-tier heading called 'Perception Step'. I recommend split E05'Color Thresholding' into two pieces 'The Perception Step' and 'Color Thresholding'.   BTW, adding a link or video to the introduction of numpy and matplotlib would be perfect.

sychaichangkun, 2017-04-26:
The font-size of the second-tier heading 'Perception Step' in E05 seems smaller the that of E06

ryan-keenan, 2017-05-02:
Thanks for all this great feedback!  I've addressed all these changes now.  Sub-heading font sizes made consistent, links added to numpy and matplotlib pages, two sub-heading now exist "Reading in Color Images" and "Color Thresholding" and the name of the page has been switched to "The Perception Step".  Also added another image above the quiz to make it clear what the output should look like.

closed 2017-05-02 by ryan-keenan