udacity/robotics-beta-test-issues #156: sagunms P1E6 Video walkthrough on Perspective transformation

sagunms, 2017-04-26:
A video demonstration similar to [CarND video](https://classroom.udacity.com/nanodegrees/nd013/parts/fbf77062-5703-404e-b60c-95b78b2f3f9e/modules/2b62a1c3-e151-4a0e-b6b6-e424fa46ceab/lessons/40ec78ee-fb7c-4b53-94a8-028c5c60b858/concepts/5aba544c-13b9-440f-b35c-f67aef7e2946) would be really good because this there is a lot of concepts going on in a single page. 

Explaining the intuition behind Perspective Transformation; about how the source coordinates for the transform are obtained via Matplotlib, along with a python notebook walkthrough would be really good and why thing would be a good basis for a map representation.

Shreeyak, 2017-04-30:
I second this. Was about to post an issue regarding the same. Some more context needs to be given about perspectives and transformation. A small video or a gif and few lines should be included at least.

ryan-keenan, 2017-05-01:
Thanks for the feedback on this!  I am working on putting something together here :).

ryan-keenan, 2017-05-22:
video now in the classroom!

closed 2017-05-22 by ryan-keenan