udacity/robotics-beta-test-issues #166: letyrodri P1

letyrodridc, 2017-04-29:
perception.py seems incorrect.

Parameters src, dst and variable img_size are not defined.

It seems like a part is missing, should be filled by the student? 

I think that img_size should be defined.

Code: img_size = (img.shape[1], img.shape[0])

Current function:
def perspect_transform(img, src, dst):
       
    M = cv2.getPerspectiveTransform(src, dst)
    warped = cv2.warpPerspective(img, M, img_size)# keep same size as input image
    return warped

letyrodridc, 2017-04-29:
Or img_size could be added as parameter too.

ryan-keenan, 2017-04-29:
Thanks for catching this!  Just pushed the fix, now the function looks like this
```
def perspect_transform(img, src, dst):
       
    M = cv2.getPerspectiveTransform(src, dst)
    warped = cv2.warpPerspective(img, M, (img.shape[1], img.shape[0]))# keep same size as input image
    return warped
```

closed 2017-04-29 by ryan-keenan