udacity/robotics-beta-test-issues #167: vivek1108: Map coordinates

vxy10, 2017-04-29:
in project, the map is loaded by reading in the map file in calibration_images folder. However, when the map is read, the images are stored as 200X200X3 nd-array with top-left as origin. It will be helpful to include a note on this, so the students are aware.

ryan-keenan, 2017-04-29:
Thanks for the suggestion Vivek, I've added some comments around this in `drive_rover.py`.  Hopefully that helps alleviate confusion, but this is definitely a tricky point that will tend to trip people up.  I'll think more about how to clarify this.

closed 2017-04-29 by ryan-keenan

vxy10, 2017-04-29:
Yeah, another confusion is that x and y are reversed in images. I still get confused by that. I implemented position of robot like this, 

`map_img[200-rover2worlds[1],rover2worlds[0],:] = [255,0,0]`

Is this correct? I did check from data by plotting on the map, and it seems to be correct. But not sure if its the right thing to do or not.