udacity/robotics-beta-test-issues #170: vivek1108: Map scale clarification

vxy10, 2017-04-29:
Not sure if i mentioned this before. The perspective transform is of size 320X160, i.e., there twice as many pixels along x than y, and from there we transform to a 200X200 world which is square. assuming same scaling along x and y. This part must be clarified or a small comment must be included here to avoid confusion.

ryan-keenan, 2017-05-01:
Actually, I'm just advocating for doing the perspective transform such that a 10x10 pixel square represents 1 square meter (could be any size image) and then mapping to a world where each pixel is 1 square meter (again could be any size world).  I'll add something to clarify this in the lesson and project

ryan-keenan, 2017-05-04:
I've added a note about this in the "warp, threshold and map" concept.  Will keep an eye out to see if this is tripping people up.

closed 2017-05-04 by ryan-keenan