udacity/robotics-beta-test-issues #171: vivek1108: vision_image and worldmap

vxy10, 2017-04-30:
I am consufed about what vision_image and worldmap variables are. Will be nice to clarify them,

ryan-keenan, 2017-05-01:
Indeed, sorry about that.  Basically these are what I'm outputting to the screen to be displayed while driving around in autonomous mode.  If you look at the section in `drive_rover.py` where you're getting ready to send commands those images get converted to base64 strings and send back to the server

```
            plotmap = np.rot90(Rover.worldmap.clip(0, 255))
            map_add = cv2.addWeighted(plotmap, 1, Rover.ground_truth, 0.4, 0)
            
            pil_img = Image.fromarray(map_add.clip(0, 255).astype(np.uint8))
            buff = BytesIO()
            pil_img.save(buff, format="JPEG")
            outImageString1 = base64.b64encode(buff.getvalue()).decode("utf-8")
            
            pil_img = Image.fromarray(Rover.vision_image.astype(np.uint8))
            buff = BytesIO()
            pil_img.save(buff, format="JPEG")
            outImageString2 = base64.b64encode(buff.getvalue()).decode("utf-8")
            
            # The action step!  Send commands to the rover!
            
            commands = (Rover.throttle, Rover.brake, Rover.steer)
            print(commands)
            send_control(commands, outImageString1, outImageString2)
```

ryan-keenan, 2017-05-01:
Will add some clarity on this going forward

ryan-keenan, 2017-05-22:
This is clarified now in comments in the project code.  Hopefully more transparent now.  I'll probably make a project walkthrough video as well.

closed 2017-05-22 by ryan-keenan