udacity/robotics-beta-test-issues #172: crl P01E04

zenetio, 2017-04-30:
Some time ago (I could not find the old ticket), I reported an issue where I said I was getting an error in notebook when trying to save the video clip to disk. In fact I find out the root cause. The reason, I was using the same directory to save the telemetry data, I mean, I just did click the r key, got the same directory and got more data. But if the directory already exists, and the log file as well, the simulator will add more images to directory and, guess what, will add more telemetry data to the same csv file, including one more header to file content. So, when I tried to process the csv file, in one point of the file the API will find a new header string, instead of a telemetry record, raising an error. So, to avoid students go into the same scenario, please advise the student that the same csv file will be used and its content will not be properly formatted (needless to say that image directory will grow as well) which will cause an error during video processing. Or you can change the code in notebook to deal with this scenario.

ryan-keenan, 2017-05-01:
Ah very interesting, I think I heard this from someone else as well.  If I set it up so that the IMG folder and csv file are just wiped out and started fresh each time does that seem like a good solution?  That way the notebook code would work well and you won't have overlapping datasets in the folder or the csv.

zenetio, 2017-05-02:
To avoid problems in notebook when reading, I used indexes like robot_log1.csv, robot_log2.csv, etc. So, I could keep any previous saved dataset available to be reviewed, if necessary, and as well avoided the crash during the clip creation.

ryan-keenan, 2017-05-02:
Ok the new version of the simulator is setup to wipe out the IMG folder and csv file.  I realize this may cause some people to lose data they might have been trying to keep, but seems like a clean solution for the moment.  I'll add a note to change folder/csv file names if you want to record more data but keep the old stuff too.

closed 2017-05-02 by ryan-keenan