udacity/robotics-beta-test-issues #183: gabrcg P01L06

gabrcg, 2017-05-04:
This doesn't work in Jupyter:

In this interactive matplotlib window you can read off the pixel coordinates of the cursor in the lower left-hand corner. You can zoom in on each corner (using the zoom tool in the upper-right) of the square to get a better measurement. Take note of the x and y coordinates of each corner.

ryan-keenan, 2017-05-05:
Thanks for pointing this out!  This works is you use `%matplotlib qt` or `%matplotlib notebook` in a notebook.  I will add this explicitly to the lesson.

ryan-keenan, 2017-05-09:
I've just recorded a video about this and added a note in the code in the lesson on how to bring up an interactive matplotlib window in the notebook.  Hopefully this helps!

closed 2017-05-09 by ryan-keenan