udacity/robotics-beta-test-issues #184: gabrcg P01

gabrcg, 2017-05-05:
In the Jupyter notebook, the block 'Make a video from processed image data' is returning the following error:

NeedDownloadError: Need ffmpeg exe. You can download it by calling:
  imageio.plugins.ffmpeg.download()

kylesf, 2017-05-06:
Just run the function `imageio.plugins.ffmpeg.download()` in order to download that software for your system. This will take care of that error.

gabrcg, 2017-05-08:
yes, it worked, but maybe it should be in the code, or in the text

import imageio
imageio.plugins.ffmpeg.download()

ryan-keenan, 2017-05-09:
Thanks for reporting this, most people don't see this error but perhaps we'll add some more instruction around this if it becomes a problem.

closed 2017-05-09 by ryan-keenan