udacity/robotics-beta-test-issues #187: Usman L02E07

UsmanIjaz, 2017-05-08:
The expected result of rover_coords() and the result from solution provided are different.

ryan-keenan, 2017-05-09:
Hmmm... should be the same.  Can you refresh the page, hit the "Reset Quiz" button and try again?

UsmanIjaz, 2017-05-09:
I checked it. The problem is with the x-axis and y-axis which are opposite in the result achieved by solution, to the image shown just above the quiz.
May be I am missing something here. But, I think the expected result is shared just above the quiz and the result gotten using the solution are different.

ryan-keenan, 2017-05-09:
can you paste the output from the quiz here?  to me it looks just like this image: 
[image]

ryan-keenan, 2017-05-09:
and be sure to hit the `RESET QUIZ` button

UsmanIjaz, 2017-05-09:
That was  strange. But the good thing is now it works. :)
It's exactly the same.

ryan-keenan, 2017-05-09:
Yay!

closed 2017-05-22 by ryan-keenan