udacity/robotics-beta-test-issues #188: letyrodri P1

letyrodridc, 2017-05-08:
Something strange is going on with coordinates in Rover project after the last update.

I'm trying to update my old working version of the project to the new one. 

The worldmap is not right and the Rover doesn't go in the direction that suppose to. I've realized that rover_coords and to_polar_coords have changed, switching the x and y.

Something could be still wrong. 

The to_polar_coords, it's alright because it's switching y and x according to arctan2 definition.

But rover_coords has something wrong. My intuition is that the older version methoed has the x and y switched. Could variables xpos and ypos inside method be wrong? 

For determining x_pixel, it uses ypos. And for determining y_pixel, it uses xpos.

letyrodridc, 2017-05-08:
My intuition is that the older version method has the x and y, already, switched. And that's why it was working with arctan2.

letyrodridc, 2017-05-08:
Update: If in method *rover_cords()*, I remove the leading minus "-" at variable ypos assigment, it works.

Changing this:
'y_pixel = -(xpos - binary_img.shape[0]).astype(np.float)'  

to this:
'y_pixel = (xpos - binary_img.shape[0]).astype(np.float)'

ryan-keenan, 2017-05-09:
Thanks for reporting this!  So, inside the simulator the definition of yaw angle has changed and that may be the problem.  Now yaw is defined counterclockwise from positive x (previously clockwise from positive y).  If you download a fresh version of the simulator (and grab the updated repository for the project) things should be working again!

ryan-keenan, 2017-05-22:
ok closing this now, if you still have any problems let me know but the current version of the project in the repo should work well with the current sim.

closed 2017-05-22 by ryan-keenan