udacity/robotics-beta-test-issues #195: sjuneja L02E05 The Perception Step

sahiljuneja, 2017-05-09:
1. Image sizes

When introducing the image size with respect to the axes (like y and x) it might possibly help if you pointed out which is referred to as "width" or "length" as well. 

This very small detail is what confuses many initially, especially when this has to be converted into code with arrays.

2. Color Thresholding

A small example of what color thresholding actually is or does before the exercise would be helpful. A step-by-step example would be better to know that the student has to define a value above which this happens, and below which something else happens for different channels, before asking them to implement the entire thing maybe? Or do you think that's a bit too much hand-holding? Seems it might help for beginners, right? Let me know what you think.

jchernus, 2017-05-09:
I agree with point #2. The Self-Driving Car ND's first lesson is a lot better at explaining this concept. Maybe we can 'borrow' some of their content from there?

ryan-keenan, 2017-05-18:
I agree this treatment of color thresholding is a little sparse.  I'm not really inclined to change it at this moment just because we have a lot of other stuff to pull together, but I'm interested in hearing whether you think the most useful addition would be a video like CarND?  Or just a better code example?  Or something else?

sahiljuneja, 2017-05-19:
Hmm.. I am thinking what's best for both you guys and the students. The thing is, there's quite a bit of text for the first week. So I am inclined towards a video because that, as per me, should ideally give a good break for the students from continuous reading. And since your goal for the first week is to engage them with the project in a slightly more superficial manner (rather than in-depth as you mentioned in another one of my posts), a small video explaining what thresholding is about would be better. Would you agree?

But not sure your timeline supports a video now. That's the only thing. This could easily be explained through words and some extra code too since that might be faster to achieve from an adding-to-the-course point of view.

ryan-keenan, 2017-05-19:
That sounds good, we can certainly add a video.  Won't be by launch next week, but soon thereafter!

ryan-keenan, 2017-05-22:
I'm going to close this now and as soon as I get a chance I'll record a video here and probably a project walkthrough as well.

closed 2017-05-22 by ryan-keenan