udacity/robotics-beta-test-issues #206: L02E07 - missing src, dest

jchernus, 2017-05-10:
In the first code block, 

warped = perspect_transform(image)

should say

warped = perspect_transform(image, source, destination)

yuvaramsingh94, 2017-05-10:
if you see the code  exercise , inside perspect_transform(img) function , both the src and dest are defined as local variables . that is why they where not send as argument

jchernus, 2017-05-11:
Oh, I see. The awkward thing here is that someone like me will be copying and pasting each piece of example code into my local environment and running it there - to make sure that I am understanding each concept and following everything completely. 

I wish they didn't change that. =/

ryan-keenan, 2017-05-19:
Hey, thanks for catching this Julia, this was an error for sure.  The page should be all updated now so that the example code and the quiz are consistent.

closed 2017-05-19 by ryan-keenan