udacity/robotics-beta-test-issues #209: Shreeyak P01E02

Shreeyak, 2017-05-10:
The "alt " command to bring up the settings doesn't work for me. I'm using Ubuntu. Also, there are screenshots for MacOS and Windows but none for Ubuntu. Tried hitting alt on startup for 5min.

ryan-keenan, 2017-05-10:
Thanks for reporting this! On windows you need to double click then hit Alt immediately... does that work for you? I'm considering dropping this "feature" and just having that screen pop up every time you launch

Shreeyak, 2017-05-11:
That did not work for me, no. I read the thread on this at slack, tried
everything. Couldn't get it working. Best would be to add an menu option in
the start screen or always have this option pop up.

ryan-keenan, 2017-05-22:
The new version of the sim brings this window up automatically, no need to hold alt down

ryan-keenan, 2017-05-22:
The new version of the sim brings this window up automatically, no need to press alt

closed 2017-05-22 by ryan-keenan