udacity/robotics-beta-test-issues #217: crl L03E02

zenetio, 2017-05-11:
In this quiz, I was looking for an option with only forward or back because there is no way to know the quantity value. But, yes, it make sense to use 40 steps in the explanation.

ryan-keenan, 2017-05-11:
Could you clarify which quiz you're referring to?  I know Mike originally said to list things like L03E02 but now that we've opened more content this is ambiguous.  Could you provide a link instead?

zenetio, 2017-05-11:
There is no number in the quiz, which could be helpful, but I am referring to the quiz at end of Turtlesim Overview.

zenetio, 2017-05-11:
https://classroom.udacity.com/nanodegrees/nd209-beta/parts/f491093d-8e67-4eab-8102-87bae095ca79/modules/2919466f-aa2b-4424-b86a-98b0a53ce335/lessons/2cd33882-d29e-43e8-9ff7-10398c8b5351/concepts/11f543ab-8ee5-421a-9e89-d67e68ba6ac9

bkinman, 2017-05-15:
Thanks for the feedback! Being multiple choice in nature, I think it's fine to leave it as is. Using just forwards and backwards, without quantifying the distance traveled would be less correct, in my opinion, as you must always specify _some_ value.

closed 2017-05-15 by bkinman