udacity/robotics-beta-test-issues #226: cornzyblack_JupyterNotebook

cornzyblack, 2017-05-11:
I once ran Jupyter notebook from Anaconda Prompt and even bash and they both worked. It doesn't seem to work again, i am always getting "failed to create process" error.

cornzyblack, 2017-05-11:
Solved this by recreating a new RoboND environment.

ryan-keenan, 2017-05-15:
Hey, sorry for the delay on this response but seems like a confusing issue... so have you ever been able to `import cv2`?  Did you install python following the [RoboND Python Starterkit](https://github.com/ryan-keenan/RoboND-Python-Starterkit)?

cornzyblack, 2017-05-16:
Yes it has worked before, even jupyter notebook has. It just gives a "failed to process" error on later runs.

ryan-keenan, 2017-05-16:
Hmmm... very strange.  It would be ideal if you could recreate the scenario where it works vs. where it fails to in order to better diagnose.  I did search for "jupyter notebook failed to create process" and found what looks like a promising thread:  https://groups.google.com/a/continuum.io/forum/#!topic/anaconda/wlibvnzd7QA
potentially one of these solutions would apply?

cornzyblack, 2017-05-16:
I have actually followed the instructions there. In fact that was the last solution I tried. Let me recreate the scenario again.

cornzyblack, 2017-05-16:
When jupyter notebook works this time, it gives another error. 
[image]
[image]

Now i tried verifying if opencv was installed, and yes it was
[image]

ryan-keenan, 2017-05-16:
I think this could be a kernel issue.  Within the notebook select the "Kernel" menu, choose "Change Kernel" and you should see one listed as "Python [conda env: RoboND]"  (or wherever you have installed `cv2`

cornzyblack, 2017-05-16:
There is no RoboND option in the kernel 
[image]

It also shows the same thing when i run **$python drive_rover.py** even when i have activated the RoboND environment

ryan-keenan, 2017-05-16:
Very strange.  Sounds like there's a mismatch between where you are running things and where Python is looking for packages.  I don't have bandwidth to dive deep on this right now but I'd recommend trying to figure out where Python is looking for packages compared to where you're installing things.  sorry!

cornzyblack, 2017-05-17:
And finally here is a picture of me running
$python drive_rover.py

It's working here but after i stop it and run jupyter notebook
i get the errors i have been complaining about 

[image]

closed 2017-05-22 by ryan-keenan