udacity/robotics-beta-test-issues #235: diyjac_P1 Simulator stuck after picking up last rock

diyjac, 2017-05-11:
After fixing `supporting_functions.py` to enable retrieval of the last rock (6th) by updating from
```
       rock_world_pos = Rover.worldmap[:,:,1].nonzero()
       # If there are, we'll step through the known sample positions
       # to confirm whether detections are real
       if rock_world_pos[0].any():
             rock_size = 2
             for idx in range(len(Rover.samples_pos[0]) - 1):
                   test_rock_x = Rover.samples_pos[0][idx]
                   test_rock_y = Rover.samples_pos[1][idx]
```
to
```
       rock_world_pos = Rover.worldmap[:,:,1].nonzero()
       # If there are, we'll step through the known sample positions
       # to confirm whether detections are real
       if rock_world_pos[0].any() and Rover.samples_pos[0].any():
             rock_size = 2
             for idx in range(len(Rover.samples_pos[0])):
                   test_rock_x = Rover.samples_pos[0][idx]
                   test_rock_y = Rover.samples_pos[1][idx]
```
The simulator gets stuck after retrieving that last rock and no longer sends any telemetry data.  We know it was still in the middle of picking up the last rock because the `data['picking_up']` still has a value of `1`.

The following shows an example where the simulator only sees 5 rocks instead of 6.
[image]

The following shows an example where the simulator gets stuck after enabling it to get the last rock:
[image]

diyjac, 2017-05-16:
I fixed the problem in the RoboND-Rover-Unity-Simulator.  There were two issues that created this situation:

1. Race condition in `RoverController.cs` where multiple `DoPickup()` were spawned, which exhausted the storedLocations array and caused the System.NullReferenceException to be thrown: "No more positions left on the rover!".
2. When forming the JSON array for the rock samples, `CommandServer.cs` removes the extra comma at the end of the JSON array; however, that should not be done if there are no rock samples left.

When these issues are fixed.  The rover no longer has issues with picking up the last rock sample (6).

[image]

ryan-keenan, 2017-05-16:
Thanks for fixing this John!

closed 2017-05-16 by ryan-keenan

diyjac, 2017-05-17:
You are welcome Ryan! Glad to help :D