udacity/robotics-beta-test-issues #251: sjuneja L02E07

sahiljuneja, 2017-05-13:
I am not sure if I mentioned this before, but do you guys plan to have more content building up on this first week?

Because there needs to be a clarification on what a world coordinate system is, what coordinate transformation is, what robot coordinate system is and related terms. 

It might get confusing for many new students when you reach kinematics, especially D-H parameters (which I think you plan to teach).

I like the approach where you let them go through this from a practical challenge. I think it would be effective. But would like to know if students will get balanced content to reduce confusion on these terms.

ryan-keenan, 2017-05-21:
Thanks for this suggestion, I've finally gotten around to thinking about it.  I've added some loose definitions of rover and world coordinate frames as well as an image that hopefully is worth a thousand words!

closed 2017-05-21 by ryan-keenan

sahiljuneja, 2017-05-21:
Yes, that helps! Thanks! Let's see how the students take it later :)