udacity/robotics-beta-test-issues #253: letyrodri P1

letyrodridc, 2017-05-13:
Data with the key near_sample is not filled supporting functions update_rover.

I could be useful to pick yellow rocks.

letyrodridc, 2017-05-13:
Also, picking up. The supporting_functions.py is not going to be submitted, so it's not available to students modification. 

The file could be add for submission to do complex project.

ryan-keenan, 2017-05-21:
Thanks for this suggestion!  I have added a "near_sample" field to the `RoverState()` object and update that field with `data["near_sample"]` in `update_rover()`.  I've added `supporting_functions.py` to the list of materials to submit because indeed I hope students will modify and improve this script to suit their needs!

closed 2017-05-21 by ryan-keenan