udacity/robotics-beta-test-issues #254: sjuneja L02E09

sahiljuneja, 2017-05-13:
- Explanation on why it's required to convert pixels to polar coordinates.

- Instead of the Warped Image, do you think it's better to show the original unwarped image? Or both of them? Wouldn't it give the student a better idea of what exactly the rover-centric coordinates relate to?

- The following sentence
```
Here's a plot of the pixel values in rover-centric coordinates like you've seen before, but this time where there's clearly more navigable terrain to the left.
```

I haven't yet tried this piece of code on my own, so the following might be incorrect. But firstly, shouldn't it be more terrain on the "right"? Also, what does this new plot represent? The plot from L02E07 showed a navigable area, and this plot is a transformed version of that area, correct? Because in the code below the image you use pixels from `rover_coords()` to convert to polar. Transformed into polar coordinate system. So now there's more navigable area because of that?

Also the "more navigable area" doesn't really match up with the world map, right? 

[Sorry if this seems like a dumbed down question. It was a bit confusion to interpret what your steps entailed]

sahiljuneja, 2017-05-13:
Oh, wait, when you said, "more navigable terrain to the left" that was with respect to the robot, not the plot we were seeing? Dunno why it took me a while to realize that.

ryan-keenan, 2017-05-21:
Good points!  I've redone this page with better images and descriptions (and dropped the angle vs. distance plot which seemed not so helpful)

closed 2017-05-21 by ryan-keenan