udacity/robotics-beta-test-issues #256: Shreeyak P01E06: ImportError

Shreeyak, 2017-05-14:
I'm running native Ubuntu 16.04 with ROS installed. Running jupyter from the miniconda env provided. Upon trying to import cv2, I get the following error: 
`ImportError: /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so: undefined symbol: PyCObject_Type`

This is solved by not sourcing ROS. Found the temp solution here: http://stackoverflow.com/questions/43019951/after-install-ros-kinetic-cannot-import-opencv

Can't figure out a better solution.

bkinman, 2017-05-15:
Can you verify that you were running anaconda's python executable, rather than /bin/python?

Shreeyak, 2017-05-15:
Which command should I run to check?

ryan-keenan, 2017-05-15:
Was just reading [here](http://stackoverflow.com/questions/17386880/does-anaconda-create-a-separate-pythonpath-variable-for-each-new-environment) and looks like although Anaconda does not use PYTHONPATH, if PYTHONPATH is set (by sourcing ROS for example) it may cause problems within your anaconda env.  Suggestion in the linked post is to `unset PYTHONPATH` before sourcing anaconda env.  Haven't tried it, but curious if this solves it!

Shreeyak, 2017-05-15:
I'll give it a shot and let you know tonight.

bkinman, 2017-05-23:
Any update here?

Shreeyak, 2017-05-23:
I simply removed the line that sourced ROS from my bashrc file to solve
this issue (not really a proper solution). I haven't tried the unset
method. I can try later if you want.

bkinman, 2017-05-24:
I think the best fix here might to be make note of this issue and solution in the first week lecture. Assigning to @ryan-keenan ... Ryan, feel free to reach out to me if you think there's a better option.

Shreeyak, 2017-05-28:
Hey, I tried the `unset PYTHONPATH` method and it worked. Here's the solution I used:
https://gitlab.com/robond-issues/project-search-and-sample-return/wikis/errors-1#error1-importerror-optroskineticlibpython27dist-packagescv2so-undefined-symbol-pycobject_type

Shreeyak, 2017-05-28:
This is a known (and ignored) issue:
https://github.com/conda/conda/issues/2590

I took my solution from here:
https://www.reddit.com/r/Python/comments/3b0stp/setting_the_correct_pythonpath/

ryan-keenan, 2017-05-30:
I now link to [a doc](https://github.com/udacity/RoboND-Python-StarterKit/blob/master/doc/linux_ros_anaconda_warning.md) in the Python Starter kit instructions as well as in the lesson to help people with this issue.  I'll close this issue for now but feel free to reopen if issues persist or this solution is not satisfactory for some reason.

closed 2017-05-30 by ryan-keenan