udacity/robotics-beta-test-issues #267: Shreeyak P01E07: plt.show() in Jupyter

Shreeyak, 2017-05-14:
In the examples provided, this is mentioned: `plt.show() # not necessary inside Jupyter notebook`
But I find that when I do not include this statement, I get in the output something like Out[70]: (0,160). Don't quite understand why that is there or what it means. 
When I include plt.show() this statement does not appear.
Screenshot below

Shreeyak, 2017-05-14:
[image]

Shreeyak, 2017-05-14:
Also get other messages, like this:
[image]

ryan-keenan, 2017-05-19:
I'm not sure why that is either

closed 2017-05-19 by ryan-keenan