udacity/robotics-beta-test-issues #269: Shreeyak P01E07: Wrong logic in code

Shreeyak, 2017-05-14:
In the solution.py, here is the code given:
```
# Calculate pixel positions with reference to the rover position being at the 
    # center bottom of the image.  
    x_pixel = np.absolute(ypos - binary_img.shape[0]).astype(np.float)
    y_pixel = -(xpos - binary_img.shape[0]).astype(np.float)
    return x_pixel, y_pixel
```

Here, while the values are correct, I believe the logic used is wrong.
1. In x_pixel, instead of using np.absolute, we should multiply by -1 to mirror.
2. In y_pixel, we are subtracting `binary_img.shape[0]` (=160) while what we should be doing is subracting 'binary_img.shape[1]' and then shift it by 160. 
   Or, since camera is in the centre, we subtract 'binary_img.shape[1]/2'

The result gotten is the same, of course, but the logic shown in the solution is misleading.

ryan-keenan, 2017-05-21:
Thanks for reporting this!  This is now fixed.  The reason for typecasting to float is just to be safe since the next operations that will be performed on these values are a rotation and translation, which should be done as floating point operations.  It may actually not matter here because yaw values come in as float and might force output to float, but anyway, I'm in the habit of converting to float whenever doing math so as not to get weird results.

closed 2017-05-21 by ryan-keenan

Shreeyak, 2017-05-21:
That's a good tip! Thanks!