udacity/robotics-beta-test-issues #270: generated video not showing in the jupyter notebook

867sec, 2017-05-14:


867sec, 2017-05-14:
Here I provide the current code in the last cell that doesn't work because it doesn;t show the generated video. Then below I provide the code tha I used to make the video show and play.

Code that doesn't work:
import os
from IPython.display import HTML
# print(output)
# print(os.path.dirname(os.path.realpath(output)))
# HTML("""
# <video width="960" height="540" controls>
#   <source src="{0} type="video/mp4"">
# </video>
# """.format(os.path.dirname(os.path.realpath(output))))

Code that works:

import io
import base64
video = io.open(output, 'r+b').read()
encoded_video = base64.b64encode(video)
HTML(data='''<video alt="test" controls>
                <source src="data:video/mp4;base64,{0}" type="video/mp4" />
             </video>'''.format(encoded_video.decode('ascii')))

ryan-keenan, 2017-05-14:
Hmmm.... yeah a couple people have reported issues with the video not playing.   I just tried your suggestion and it didn't work for me, but the default code provided in the notebook does work for me.  Not sure what the problem is here.

867sec, 2017-05-15:
My suggestion didn't work for you because I didn't wrap the code properly and just noticed that in my description the waffle interface removed the HTML code between double quotes assigned to data. Please try again with this code:
```python
import io
import base64
video = io.open(output, 'r+b').read()
encoded_video = base64.b64encode(video)
HTML(data='''<video alt="test" controls>
                <source src="data:video/mp4;base64,{0}" type="video/mp4" />
             </video>'''.format(encoded_video.decode('ascii')))
```

ryan-keenan, 2017-05-15:
Ok thanks, that works for me so I've just added it to the bottom of the notebook as an alternative video viewing method.  Don't want to remove the other one just yet since it works for most people.

closed 2017-05-15 by ryan-keenan