udacity/robotics-beta-test-issues #280: abhi_ L04E04

abhio9vt, 2017-05-15:
the command says: simple_arm robot_spawn.launch
it should be: roslaunch simple_arm robot_spawn.launch

867sec, 2017-05-16:
I agree something is wrong here. Did you get it to work? What folder should we execute this command from? I'm getting the following error when I run your command "roslaunch simple_arm robot_spawn.launch":
"[robot_spawn.launch] is neither a launch file in package [simple_arm] nor is [simple_arm] a launch file name"

abhio9vt, 2017-05-16:
run make command on your catkin workspace.

867sec, 2017-05-16:
I did, previously, but didn't again just now and still the same error...
Can you help?
robo@robo-virtual-machine:~/catkin_ws$ catkin_make
Base path: /home/robo/catkin_ws
Source space: /home/robo/catkin_ws/src
Build space: /home/robo/catkin_ws/build
Devel space: /home/robo/catkin_ws/devel
Install space: /home/robo/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in
"/home/robo/catkin_ws/build"
####
####
#### Running command: "make -j2 -l2" in "/home/robo/catkin_ws/build"
####
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target _simple_arm_generate_messages_check_deps_GoToPosition
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_py
[ 14%] Built target simple_arm_generate_messages_lisp
[ 42%] Built target simple_arm_generate_messages_eus
[ 71%] Built target simple_arm_generate_messages_nodejs
[ 71%] Built target simple_arm_generate_messages_cpp
[100%] Built target simple_arm_generate_messages_py
[100%] Built target simple_arm_generate_messages
robo@robo-virtual-machine:~/catkin_ws$ roslaunch simple_arm
robot_spawn.launch
[robot_spawn.launch] is neither a launch file in package [simple_arm] nor
is [simple_arm] a launch file name
The traceback for the exception was written to the log file
robo@robo-virtual-machine:~/catkin_ws$

bkinman, 2017-05-23:
> What folder should we execute this command from? 
Anywhere, as long as `setup.bash` has been sourced from ~/catkin_ws/devel

Perhaps this is the problem you are running into? I'm going to assume it was and close this issue. Please reopen it if you are still having the problem.

closed 2017-05-23 by bkinman