udacity/robotics-beta-test-issues #283: Vivek Yadav: Rover.near_sample is not set to true

vxy10, 2017-05-15:
Hi,
I implemented code to spot for yellow samples, but when the rover went close to samples, the near_sample variable was not set to true in python file, but was set to true in simulator. Does this need to be fixed?

vxy10, 2017-05-15:
vivek1108: Here is the image, https://drive.google.com/file/d/0B1fs4jieX9OOQTRQZjZxcm9JQ3c/view?usp=sharing

ryan-keenan, 2017-05-22:
I was just trying this out today and when I'm near a sample I get back a 1 in data["near_sample"] from telemetry.  Are you still having this problem?  I've updated `RoverState()` now to have a `Rover.near_sample` field updated in `update_rover()` and it seems to work.

vxy10, 2017-05-22:
No, i was looking at Rover.near_sample, not data['near_sample']. data['near_sample'] was 1

closed 2017-05-22 by ryan-keenan