udacity/robotics-beta-test-issues #286: sagun P01 Simulator implementation of "fixed_turn" command doesn't work

sagunms, 2017-05-15:
I noticed that the simulator can react to a "fixed_turn" command over socket.io.
```python
def send_fixed_turned(angle, time):
    data = {
        'angle': angle.__str__(),
        'time': time.__str__(), 
        'throttle': (1.0).__str__(),
    }
    sio.emit(
        "fixed_turn",
        data,
        skip_sid=True)
```
Executing the following command did not do a fixed turn:

```python
do_fixed_turn = {'angle': 10, 'time': 5.0}
if (do_fixed_turn['angle'] * do_fixed_turn['time']) != 0:
    send_fixed_turned(do_fixed_turn['angle'], 
                      do_fixed_turn['time'])
```

Or am I doing something wrong?

sagunms, 2017-05-15:
I think I used radians:
```python
do_fixed_turn = {'angle': direction * np.pi / 2, 'time': 5.0}
```

I think it should be included as part of course instructions.

ryan-keenan, 2017-05-22:
I've added the "pickup" instructions [here](https://classroom.udacity.com/nanodegrees/nd209-beta/parts/f491093d-8e67-4eab-8102-87bae095ca79/modules/0573f9ff-2e23-48cd-87d0-af24d758348a/lessons/ef841d31-8c53-49b3-8da3-a9d1523adef0/concepts/ed33ab6f-2b67-4991-b372-ab915538b734) and code to the scripts provided. 

I'm thinking of just removing the "fixed_turn" stuff because it doesn't seem necessary.  Do you find it useful?

sagunms, 2017-05-27:
I don't really find it necessary. I'd lean towards removing it to keep things simple. It was just an undocumented part I saw in the simulator code and was unsuccessful using it. 

The "pickup" instructions you added looks really good.

kylesf, 2017-06-10:
Is this matter deemed as resolved?

ryan-keenan, 2017-06-10:
ah yeah, fixed turn is removed now.

closed 2017-06-10 by ryan-keenan