udacity/robotics-beta-test-issues #289: diyjac L04E03

diyjac, 2017-05-15:
When trying to execute the git clone for the simple_arm as directed in the lesson, I get an access rights error in the VM.  Perhaps use the https version instead of the ssh?

```
git clone git@github.com:udacity/simple_arm_01.git simple_arn
```

867sec, 2017-05-16:
You code is http authentication (you earlier saved your password in git cache), the course code is for ssh authentication, when you need to use ssh key saved in git cache. Both are good, and I agree that both methods should be mentioned.

diyjac, 2017-05-16:
This is specifically for the VM supplied by Udacity.  While it maybe possible for the student to customize the robo account to their own github later on for a project, it does not seem likely at the beginning of the lesson.

867sec, 2017-05-16:
I see, yes, you are right, that makes sense.

bkinman, 2017-05-23:
I've switched to https, this should fix the problem. Students will need to have an account on github though.

closed 2017-05-23 by bkinman