udacity/robotics-beta-test-issues #293: diyjac L04E05

diyjac, 2017-05-15:
After installing the dependencies, I am having problems with keeping the launched process up.  Here are the errors:

```
robo@robo-virtual-machine:~/catkin_ws$ roslaunch simple_arm robot_spawn.launch
... logging to /home/robo/.ros/log/b231bb9a-3971-11e7-a254-000c296298c0/roslaunch-robo-virtual-machine-2423.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://robo-virtual-machine:43029/

SUMMARY
========

PARAMETERS
 * /joint_state_publisher/use_gui: False
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /simple_arm/joint_1_position_controller/joint: joint_1
 * /simple_arm/joint_1_position_controller/pid/d: 10.0
 * /simple_arm/joint_1_position_controller/pid/i: 0.01
 * /simple_arm/joint_1_position_controller/pid/p: 100.0
 * /simple_arm/joint_1_position_controller/type: effort_controller...
 * /simple_arm/joint_2_position_controller/joint: joint_2
 * /simple_arm/joint_2_position_controller/pid/d: 10.0
 * /simple_arm/joint_2_position_controller/pid/i: 0.01
 * /simple_arm/joint_2_position_controller/pid/p: 100.0
 * /simple_arm/joint_2_position_controller/type: effort_controller...
 * /simple_arm/joint_state_controller/publish_rate: 50
 * /simple_arm/joint_state_controller/type: joint_state_contr...
 * /use_sim_time: True

NODES
  /simple_arm/
    spawner (controller_manager/spawner)
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    urdf_spawner (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [2438]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b231bb9a-3971-11e7-a254-000c296298c0
process[rosout-1]: started with pid [2451]
started core service [/rosout]
process[joint_state_publisher-2]: started with pid [2454]
process[robot_state_publisher-3]: started with pid [2455]
process[simple_arm/spawner-4]: started with pid [2456]
process[gazebo-5]: started with pid [2474]
process[gazebo_gui-6]: started with pid [2483]
process[urdf_spawner-7]: started with pid [2488]
SpawnModel script started
[ INFO] [1494854615.629073801]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1494854615.630676703]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[INFO] [1494854615.735337, 0.000000]: Loading model XML from ros parameter
[INFO] [1494854615.737646, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1494854615.826626, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1494854616.270166133, 261.200000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1494854616.306891405, 261.223000000]: Physics dynamic reconfigure ready.
[INFO] [1494854616.341722, 261.257000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1494854616.509013, 261.366000]: Spawn status: SpawnModel: Successfully spawned model
[ INFO] [1494854616.727894235, 261.366000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[urdf_spawner-7] process has finished cleanly
log file: /home/robo/.ros/log/b231bb9a-3971-11e7-a254-000c296298c0/urdf_spawner-7*.log
[ INFO] [1494854616.741949808, 261.366000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1494854616.843240697, 261.366000000]: Loading gazebo_ros_control plugin
[ INFO] [1494854616.844887883, 261.366000000]: Starting gazebo_ros_control plugin in namespace: /simple_arm
[ INFO] [1494854616.848501906, 261.366000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[ WARN] [1494854617.027375880, 261.366000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'EffortJointInterface' within the <hardwareInterface> tag in joint 'joint_1'.
[ WARN] [1494854617.030532546, 261.366000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'EffortJointInterface' within the <hardwareInterface> tag in joint 'joint_2'.
[ INFO] [1494854617.045622868, 261.366000000]: Loaded gazebo_ros_control.
[INFO] [1494854617.335297, 261.654000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1494854617.337476, 261.656000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1494854617.339384, 261.656000]: Loading controller: joint_state_controller
[INFO] [1494854617.544413, 261.860000]: Loading controller: joint_1_position_controller
gzserver: symbol lookup error: /opt/ros/kinetic/lib//libeffort_controllers.so: undefined symbol: _ZN4urdf5Model23initParamWithNodeHandleERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKN3ros10NodeHandleE
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 212, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 190, in main
    resp = load_controller(name)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 525, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
[INFO] [1494854617.749805, 262.056000]: Shutting down spawner. Stopping and unloading controllers...
[gazebo-5] process has died [pid 2474, exit code 127, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/robo/catkin_ws/src/simple_arm/worlds/willow_garage.world __name:=gazebo __log:=/home/robo/.ros/log/b231bb9a-3971-11e7-a254-000c296298c0/gazebo-5.log].
log file: /home/robo/.ros/log/b231bb9a-3971-11e7-a254-000c296298c0/gazebo-5*.log
[WARN] [1494854648.750889, 262.056000]: Controller Spawner couldn't reach controller_manager to take down controllers. Waited for 30 second(s).
[simple_arm/spawner-4] process has died [pid 2456, exit code 1, cmd /opt/ros/kinetic/lib/controller_manager/spawner joint_state_controller joint_1_position_controller joint_2_position_controller __name:=spawner __log:=/home/robo/.ros/log/b231bb9a-3971-11e7-a254-000c296298c0/simple_arm-spawner-4.log].
log file: /home/robo/.ros/log/b231bb9a-3971-11e7-a254-000c296298c0/simple_arm-spawner-4*.log
```

bkinman, 2017-05-23:
Can you paste me the result of `dpkg --list`?

diyjac, 2017-05-23:
I think this was because the VM needed a software update.  You can see the VM being updated in this video: https://files.slack.com/files-pri/T4ET9BQ3U-F5GV58X6K/download/vm-software-update.mp4.  Once that was completed, everything worked except that gazebo needed to upgrade to 7.7 to handle VMWare not giving the correct Z value (the camera would not function and would crash gazebo_ros node).  Once that was done, the arm_mover worked properly: https://files.slack.com/files-pri/T4ET9BQ3U-F5G7ZT2JZ/vm-upgrade-to-gazebo-7.7-fixed-camera.mp4.  We can close this.

closed 2017-05-23 by diyjac