udacity/robotics-beta-test-issues #297: diyjac L04E04

diyjac, 2017-05-15:
Many command formatting issues here.  This command
```
cd ~/catkin_ws/src/simple_arm/ mkdir scripts touch GoToPosition.srv
```
Should be change to this, since `scripts` directory should have been already created previously.
```
cd ~/catkin_ws/src/simple_arm/scripts
touch GoToPosition.srv
```

alvarovillena, 2017-05-16:
I agree, but the GoToPosition.srv file must be created in the srv folder instead of in the scripts folder, because it's a service and not a python script. In addition, this service is already in the package, so there is no need to create it.

diyjac, 2017-05-16:
Good to know!  I would not have figured that out by the previous instructions.

bkinman, 2017-05-23:
In the future, please try to create separate issues for each concern.

> Many command formatting issues here.
Fixed!

> Should be change to this, since srv directory should be created instead.
Fixed!

>Although there appears to be a srv and srv/GotoPosition.srv file already there from the simple_arm git repository
Fixed!

> Also, there is a controller_manager in the CMakefile.txt. Should this entry be deleted or should the lesson be changed to reflect that controller_manager is required for arm_mover to function?

In actuality, `controller_manager` is simply a runtime dependency. The only reason it's in CMakeLists.txt is that it forces the build to break the very first time you build the package (in order to illustrate what a build breaking due to a missing dep looks like). I've added a note to students indicating that it is not required.

> The following two sections were not commented out as suggested by the lesson
Oops! Fixed!

> Both message _generation and message_runtime dependencies were already included in package.xml.
Fixed!

> Updated command formatting:
Fixed!
> and
Fixed!

Good work here!

closed 2017-05-23 by bkinman