udacity/robotics-beta-test-issues #299: L03E03 - Wrong instructions to load the .vmdk file

alvarovillena, 2017-05-15:
After downloading and installing VMware Workstation player, the lesson provides a link to load the .vmdk file and bring up the VM (https://kb.vmware.com/selfservice/microsites/search.do?language=en_US&cmd=displayKC&externalId=2010196).
However, the instructions on this link doesn't match with the options I have on my program (VMware® Workstation Player 12.5.5 on Ubuntu).
I have found two ways to bring up the virtual machine successfully.

First Method:
1. File -> Open a virtual machine
2. Select the provided .vmx file

Second Method:
1. File -> Create a new virtual machine
2. Select 'I will install the operating system later'.
3. Select 'Linux' version 'Ubuntu 64-bit' as guest OS.
4. Create the machine. Hardware settings can be customized.
5. Once the VM is created, select 'Edit virtual machine settings'. Remove the Hard Disk device and add a new Hard Disk device, using an existing virtual disk. Select the .vmdk as the disk file.

alvarovillena, 2017-05-15:
Sorry, first method steps are not complete in my first post.

First Method:
1. File -> Open a virtual machine
2. Select the provided .vmx file
3. When the VM is started for the first time, a dialog box asks us to browse for the .vmdk file. Select the provided .vmdk file.
4. The program asks if the machine has been moved or copied. Select 'I Copied It'

Shreeyak, 2017-05-28:
Hi, these instructions haven't been modified in the lesson yet, right? Had a question regarding the same from a student...

kylesf, 2017-06-13:
Fixed and new instructions can be found in the classroom.

closed 2017-06-13 by kylesf