udacity/robotics-beta-test-issues #300: Suggestion: VM keyboard layout

alvarovillena, 2017-05-15:
The default keyboard layout for the virtual machine is 'us'. This can be very annoying when working with a different layout (I always push every key until I find out where the '/' or the '-' is :D). Maybe providing some instructions to change this can be useful.

To change the keyboard layout, I've modified /etc/default/keyboard. In my case, I changed the XKBLAYOUT parameter from "en" to "es". After a reboot, the keyboard layout is changed.

Important: The password contains a hyphen. This can be hard to write with an incorrect keyboard layout. Modifying /etc/default/keyboard requires sudo, so I recommend to copy and paste the password.

Shreeyak, 2017-05-22:
It would be nice to remove the hyphen from the password. If it need be more secure, then a longer phrase can be chosen. At boot up, one can't even type stuff to see what key prints what.

kylesf, 2017-06-10:
All of this is taken care of in the new VM update. Will close this when new VM ships.

kylesf, 2017-06-12:
This has been taken care of and you can find the updated VM in the classroom! Thanks for the help!

closed 2017-06-12 by kylesf