udacity/robotics-beta-test-issues #303: L03E06 - rosed command

alvarovillena, 2017-05-15:
From the lesson: 'One way to get more detail would be to look at the comments in the message’s definition file. To do so, we can issue the following command: rosed geometry_msgs Twist.msg.'

The rosed command is a quick and easy way to open a ROS file in a text editor. Maybe a short explanation about this command is useful. In addition, using this command in the VM leads to an error because vim is not installed. The quick way to fix this is installing vim, but it may be useful to indicate that the file is opened in the editor specified in the $EDITOR env variable. Setting this variable to your favorite text editor can be useful. (yes, I've spent a lot of time trying to figure out how to close vim :P)

bkinman, 2017-05-19:
A little more explanation of what ROSED is would indeed be good. I'll close this issue when the script has been updated.

bkinman, 2017-05-23:
I've updated the text atom to include a link to the ROS wiki, where you can get more information about rosed, including setting the EDITOR environment variable.

closed 2017-05-23 by bkinman