udacity/robotics-beta-test-issues #307: L05E05 joint_state information

thiagodefreitas, 2017-05-15:
It is not completely clear to me the didactic benefit of having a rospy.wait_for_message for the Joint states, instead of having for example a Subscriber for the '/simple_arm/joint_states' message, where this information would be updated(a little bit more similar to what is done on real systems)

bkinman, 2017-05-19:
`move_arm()` must return for the response to get sent back, as the move_arm is a service request handler; Thus I'm not able to wait for a msg callback, I'm forced to use the blocking function call `wait_for_message()`.

To be honest, `wait_for_message() `is frowned upon, and using it implies that you are probably using what would be considered an anti-pattern, or bad form in ROS. Service responses should be able to be provided immediately, or close it it... I think that the `move_arm` should be an action server instead of a service. It would avoid the `wait_for_message()`. The reason that I didn't make it an action server, is that I could not imagine what an alternate service request would look like for such a simple project.

I finally came up with an idea for the service that students could implement in the scenario where move_arm() was an action... It's on a post-it note on my desk.

I'll close this issue when I've updated the code and script accordingly. It had not initially been my intention to talk about action servers; I'll have to figure out how to fit that in.

Thanks for putting in the effort to explain such an in-depth concern to me. I was already on the fence about changing this.

bkinman, 2017-05-19:
`move_arm()` is a service handler, and sends the response message upon it's return.

I can't use a message callback for `/simple_arm/joint_states`, because I'd have to do it outside of the `move_arm()` callback. This is why I elected to use `rospy.wait_for_message()`, as it's a blocking call.

The reality of the situation is that `.wait_for_message()` is frowned upon and probably should not be used, It implies bad form of the python script, as it's kind of an anti-pattern. Nothing else in the node can happen if you're sitting there blocked and waiting. Instead, the `move_arm` functionality should have been an action request instead of a service. Action server responses are meant to be sent back upon completion of an action; which by design is more or less meant to be a long-lived request-response pair.

I need to make `move_arm` an action request rather than a service.
I need to find some other example of a service to use in the place of `move_arm`.
I need to update the script to account for these changes, and to introduce students to the idea of action servers somewhere.

Keeping this open until I do the above.

Thanks for taking the time to express your concern here, as it has pushed me over the edge to do something about this.

bkinman, 2017-06-30:
Adding this to the backlog.

closed 2018-05-03 by jessluls